RobotWebTools/ros2-web-bridge #172: Running the example does not work

SomaGallai, 2021-01-26:
My setup is the following:

-     Node.js: v10.15.2
-     ROS 2 Foxy Fitzroy Latest Release
-     Ubuntu 20.04
-     npm v6.4.1

Running `npm install` does not throw any errors. ROS2 has been sourced from the start. 
Running `node bin/rosbridge.js` gives the following error:
```
/home/user/ros2-web-bridge/node_modules/rclnodejs/lib/context.js:47
  static _instances = [];
                    ^

SyntaxError: Unexpected token =
    at new Script (vm.js:79:7)
    at createScript (vm.js:251:10)
    at Object.runInThisContext (vm.js:303:10)
    at Module._compile (internal/modules/cjs/loader.js:657:28)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:22:18)
```

minggangw, 2021-01-27:
Please upgrade your nodejs to v12.20.1, meanwhile, we have fixed the [support](https://github.com/RobotWebTools/rclnodejs/commit/b6f2f0dcfed229954d5db5131a1f549603568a26) for nodejs 10.x and it will get supported in the next release.

SomaGallai, 2021-01-27:
I have upgraded nodejs to v12.20.1 after that when running `node bin/rosbridge.js` the following error occured:
```
internal/modules/cjs/loader.js:258
    throw e;
    ^

SyntaxError: Error parsing /home/user/ros2-web-bridge/package.json: Unexpected token : in JSON at position 219
    at parse (<anonymous>)
    at readPackage (internal/modules/cjs/loader.js:245:20)
    at readPackageScope (internal/modules/cjs/loader.js:270:19)
    at shouldUseESMLoader (internal/modules/run_main.js:36:15)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:55:24)
    at internal/main/run_main_module.js:17:47 {
  path: '/home/user/ros2-web-bridge/package.json'
}
```

So I have deleted the `node_modules and package-lock.json` to do a fresh install, but then the following error happened:
```
npm ERR! code EJSONPARSE
npm ERR! file /home/user/ros2-web-bridge/package.json
npm ERR! JSON.parse Failed to parse json
npm ERR! JSON.parse Unexpected token : in JSON at position 219 while parsing near '...cl",
npm ERR! JSON.parse     "rclnodejs": "0.14.0"
npm ERR! JSON.parse   ],
npm ERR! JSON.parse   "b...'
npm ERR! JSON.parse Failed to parse package.json data.
npm ERR! JSON.parse package.json must be actual JSON, not just JavaScript.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/user/.npm/_logs/2021-01-27T08_55_06_581Z-debug.log
```

SomaGallai, 2021-01-27:
I was able to fix this error by editing package.json to not specify the rclnodejs version.  `node bin/rosbridge.js and cd examples && node index.js` is still not working as expected.
 `node bin/rosbridge.js` has the following output:
```
/home/user/ros2-web-bridge/node_modules/rclnodejs/lib/context.js:47
  static _instances = [];
                    ^

SyntaxError: Unexpected token =
    at new Script (vm.js:79:7)
    at createScript (vm.js:251:10)
    at Object.runInThisContext (vm.js:303:10)
    at Module._compile (internal/modules/cjs/loader.js:657:28)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:22:18)
```
just as when I opened the issue.

SomaGallai, 2021-01-27:
I have realized for some reason it kept changing back to the older version of Node.js. When keeping an eye out it was installed with the right version and running on the good version og Node.js it is working. Thank you for your help!

closed 2021-01-27 by SomaGallai

minggangw, 2021-01-27:
Good to know the problem was gone.