RobotWebTools/ros2-web-bridge #173: Ros2webbridge crashing after a period of time

slogan-oxts, 2021-01-27:
OS: Windows 10
Ros: Ros2 Eloquent
Unity: 2019.4.14f1
Node.js: 14.15.4

I've been trying to get this working as part of [RosSharp](https://github.com/siemens/ros-sharp) but I am running into an issue where the server crashing after a period of time:

```
#
# Fatal error in , line 0
# Check failed: result.second.
#
#
#
#FailureMessage Object: 000000CFCB4FD850
 1: 00007FF7314F058F napi_wrap+109311
 2: 00007FF731424D8F std::basic_ostream<char,std::char_traits<char> >::operator<<+57151
 3: 00007FF73205E412 V8_Fatal+162
 4: 00007FF731B022AD v8::internal::BackingStore::Reallocate+653
 5: 00007FF731D480E9 v8::ArrayBuffer::GetBackingStore+137
 6: 00007FF7314D2469 napi_get_typedarray_info+393
 7: 00007FF9FA289B41
 8: 00007FF9FA28F28D
 9: 00007FF9FA287EC3
10: 00007FF9FA28EAE3
11: 00007FF7314CCD86 node::Stop+35046
12: 00007FF731D0E72F v8::internal::Builtins::builtin_handle+321471
13: 00007FF731D0DCC4 v8::internal::Builtins::builtin_handle+318804
14: 00007FF731D0DFB7 v8::internal::Builtins::builtin_handle+319559
15: 00007FF731D0DE03 v8::internal::Builtins::builtin_handle+319123
16: 00007FF731DE9F9D v8::internal::SetupIsolateDelegate::SetupHeap+464173
17: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
18: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
19: 00000267A1710CFC
```

or sometimes:
```
#
# Fatal error in , line 0
# Check failed: result.second.
#
#
#
#FailureMessage Object: 000000ADA7CFDE50
 1: 00007FF7314F058F napi_wrap+109311
 2: 00007FF731424D8F std::basic_ostream<char,std::char_traits<char> >::operator<<+57151
 3: 00007FF73205E412 V8_Fatal+162
 4: 00007FF731B022AD v8::internal::BackingStore::Reallocate+653
 5: 00007FF731D480E9 v8::ArrayBuffer::GetBackingStore+137
 6: 00007FF7314D2469 napi_get_typedarray_info+393
 7: 00007FF9FDD89B41
 8: 00007FF9FDD8F28D
 9: 00007FF9FDD87EC3
10: 00007FF9FDD8EAE3
11: 00007FF7314CCD86 node::Stop+35046
12: 00007FF731D0E72F v8::internal::Builtins::builtin_handle+321471
13: 00007FF731D0DCC4 v8::internal::Builtins::builtin_handle+318804
14: 00007FF731D0DFB7 v8::internal::Builtins::builtin_handle+319559
15: 00007FF731D0DE03 v8::internal::Builtins::builtin_handle+319123
16: 00007FF731DE9F9D v8::internal::SetupIsolateDelegate::SetupHeap+464173
17: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
18: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
19: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
20: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
21: 00007FF731DBB475 v8::internal::SetupIsolateDelegate::SetupHeap+272901
22: 00007FF731E5FEFE v8::internal::SetupIsolateDelegate::SetupHeap+947342
23: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
24: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
25: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
26: 00007FF731D7E661 v8::internal::SetupIsolateDelegate::SetupHeap+23537
27: 00007FF731E68343 v8::internal::SetupIsolateDelegate::SetupHeap+981203
28: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
29: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
30: 00007FF731D7E661 v8::internal::SetupIsolateDelegate::SetupHeap+23537
31: 00007FF731E68343 v8::internal::SetupIsolateDelegate::SetupHeap+981203
32: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
33: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
34: 00007FF731DFC0BE v8::internal::SetupIsolateDelegate::SetupHeap+538190
35: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
36: 00007FF731D8055E v8::internal::SetupIsolateDelegate::SetupHeap+31470
37: 00007FF731D8014C v8::internal::SetupIsolateDelegate::SetupHeap+30428
38: 00007FF731C50631 v8::internal::Execution::CallWasm+1649
39: 00007FF731C4FE9F v8::internal::Execution::Call+191
40: 00007FF731D3B7E7 v8::Function::Call+615
41: 00007FF7315145FB node::CallbackScope::~CallbackScope+1659
42: 00007FF731514A61 node::MakeCallback+209
43: 00007FF731514BF8 node::MakeCallback+312
44: 00007FF73151496B node::MakeCallback+139
45: 00007FF9ECA47FDD rclnodejs::ShadowNode::Execute+557 [P:\ros2webbridge\ros2-web-bridge\node_modules\rclnodejs\src\shadow_node.cpp]:L123
46: 00007FF9ECA1D59E rclnodejs::Executor::ExecuteReadyHandles+158 [P:\ros2webbridge\ros2-web-bridge\node_modules\rclnodejs\src\executor.cpp]:L258
47: 00007FF73154484B uv_async_send+331
48: 00007FF731543FDC uv_loop_init+1292
49: 00007FF73154417A uv_run+202
50: 00007FF7314502B5 v8::internal::OrderedHashTable<v8::internal::OrderedHashSet,1>::NumberOfBucketsOffset+9365
51: 00007FF7314C3A17 node::Start+311
52: 00007FF73132686C RC4_options+339820
53: 00007FF7322C604C v8::internal::compiler::RepresentationChanger::Uint32OverflowOperatorFor+153532
54: 00007FFA5C1E7C24 BaseThreadInitThunk+20
55: 00007FFA5CF8D4D1 RtlUserThreadStart+33
```
I created a simple publisher in python using rclpy which publishes a string "/topic" at 100hz.

I edited the example subscription.html to subscribe to this topic.  It connects shows the messages then after a few seconds the server crashes with one of the above errors.

minggangw, 2021-01-27:
> Node.js: 14.15.4

The crash that happens on nodejs 14.x is a known [issue](https://github.com/RobotWebTools/rclnodejs/issues/717), and it was caused by the dependency of rclnodejs, [ref-napi](https://github.com/node-ffi-napi/ref-napi/issues/47). We haven't fixed it yet.

> While testing with RosSharp, I tested an older version of Node.js 12.19.0, 

The crash that happens on nodejs 12.19.0 was caused by nodejs, please see [here](https://github.com/RobotWebTools/rclnodejs/issues/713) and it's fixed by the higher version of nodejs.

So, the solution is to use the latest Erbium nodejs, [v12.20.1](https://nodejs.org/download/release/latest-erbium/)

slogan-oxts, 2021-01-27:
Hi @minggangw 

Thanks for the quick reply.  That seems to be working perfectly now.  I would suggest this information should be put in the README.md as currently there is no indication of which version of node.js is required:

> 2.Install Node.js You can install Node.js:
> 
> Download from Node.js offical website, and install it.
> Use the Node Version Manager (nvm) to install it.

closed 2021-01-27 by slogan-oxts

minggangw, 2021-01-27:
Thanks for your suggestion, we have put the requirement of nodejs into [rclnodejs README](https://github.com/RobotWebTools/rclnodejs#prerequisites) and will add it for the ros2-web-bridge, thanks!