RobotWebTools/ros2-web-bridge #180: Is it possible to publish msg from a website to a specific robot using this ros2-web-bridge?

Hiroaki-K4, 2021-03-02:
Hi, I have a question.
Is it possible to publish msg from a website to a specific robot using this ros2-web-bridge?
Thanks!

minggangw, 2021-03-03:
I'm afraid you cannot send a topic to a specific subscriber (robot) by the current ros2-web-bridge. Will `Hidden Topic or Service Names` introduced in [topic_and_service_names](https://design.ros2.org/articles/topic_and_service_names.html) help you?

Hiroaki-K4, 2021-03-03:
@minggangw 
Thank you for reply.
What are the uses of row-web-bridge?
Does it play a role in connecting the localhost server and ros2 on the same device?

minggangw, 2021-03-03:
`row-web-bridge` can run on an independent device and the web server connects to it through WebSocket, the ros2 can run on another device, neither the `row-web-bridge` nor the web server. Normally, to simply the implementation, you can install `row-web-bridge` and ros2 on the same machine.

newcanopies, 2021-04-21:
hi @minggangw 

The below code publishes the Message values from Web-client to Gazebo, 
the inputed values do get logged in terminal as a correct JSON string, but Gazebo sim does not change -- anything missing?

1.
ros2-web-bridge Web-client app is running

2.
Web-client code:
<details>
`
 

      <div class="controllerinput">
        <div>
            <form id="ctrlPanel">
            
                <input class="crs" type="text" id="PositionX" name="PositionX" step="any" placeholder="x-coordinate" value=""></br>
            
                <input class="crs" type="text" id="PositionY" name="PositionY" step="any" placeholder="y-coordinate" value=""></br>
            
                <input class="crs" type="text" id="PositionZ" name="PositionZ" step="any" placeholder="z-coordinate" value=""></br>
            
                <input class="crs" type="text" id="OrientationR" name="OrientationR" step="any" placeholder="roll" value=""></br>
            
                <input class="crs" type="text" id="OrientationP" name="OrientationP" step="any" placeholder="pitch" value=""></br>
            
                <input class="crs" type="text" id="OrientationY" name="OrientationY" step="any" placeholder="yaw" value=""></br>
                <button id="sendMsg" type="button" class="sbtn" onclick="pubMessage(PoseStamped)">Send Coordinates</button>

            </form>
        </div>
`
</details>

Publisher script
<details>
`
<script>
        let ros = new ROSLIB.Ros({
          url : 'ws://localhost:8080', 
        });
        let bridgeHost = window.location.hostname;
        let bridgePort =8080;

        ros.on('connection', function() {
          console.log('Connected to websocket server - bridge connected.');
        });
        ros.on('error', function(data) {
          console.log('Error connecting to websocket server: ', data);
        });
        ros.on('closed', function() {
          console.log('Connection to websocket server closed - bridge closed.');
        });
        ros.connect('ws://' + bridgeHost + ':' + bridgePort);
          console.log(bridgeHost, bridgePort);
          
        var goal = new ROSLIB.Topic({
            ros : ros,
            name : '/move_base_simple/goal',
            messageType : 'geometry_msgs/PoseStamped',
        });

        var PoseStamped = new ROSLIB.Message({
            header: {
              seq: 0,
              stamp: {
                  secs: 0,
                  nsecs: 100
              },
              frame_id: "map"
            },
            pose: {
                position: {
                    x: 0.0,
                    y: 0.0,
                    z: 0.0
                },
                orientation: {
                    x: 0.0,
                    y: 0.0,
                    z: 0.0,
                    w: 1.0
                }
            }
        });
       //goal.publish(PoseStamped);

        //---------------------------------------PUBLISH
        function pubMessage() {
            var posePositionX = 0.0;
            var posePositionY = 0.0;
            var posePositionZ = 0.0;

            var poseOrientationX = 0.0;
            var poseOrientationY = 0.0;
            var poseOrientationZ = 0.0;
            var poseOrientationW = 0.0;

            // get values from HTML input
            posePositionX = Number(document.getElementById('PositionX').value);
              console.log(parseFloat(document.getElementById('PositionX').value));
            posePositionY = Number(document.getElementById('PositionY').value);
              console.log(parseFloat(document.getElementById('PositionY').value));
            posePositionZ = Number(document.getElementById('PositionZ').value);
              console.log(parseFloat(document.getElementById('PositionZ').value));
            poseOrientationR = Number(document.getElementById('OrientationR').value);
              console.log(parseFloat(document.getElementById('OrientationR').value));
            poseOrientationP = Number(document.getElementById('OrientationP').value);
              console.log(parseFloat(document.getElementById('OrientationP').value));
            poseOrientationY = Number(document.getElementById('OrientationY').value);
              console.log(parseFloat(document.getElementById('OrientationY').value));

            // Convert HTML input
            var roll = parseFloat(poseOrientationR);
            var pitch = parseFloat(poseOrientationP);
            var yaw = parseFloat(poseOrientationY);
            var phi = roll / 2.0;
            var the = pitch / 2.0;
            var psi = yaw / 2.0;
            var x = Math.sin(phi) * Math.cos(the) * Math.cos(psi) - Math.cos(phi) * Math.sin(the)
                * Math.sin(psi);
            var y = Math.cos(phi) * Math.sin(the) * Math.cos(psi) + Math.sin(phi) * Math.cos(the)
                * Math.sin(psi);
            var z = Math.cos(phi) * Math.cos(the) * Math.sin(psi) - Math.sin(phi) * Math.sin(the)
                * Math.cos(psi);
            var w = Math.cos(phi) * Math.cos(the) * Math.cos(psi) + Math.sin(phi) * Math.sin(the)
                * Math.sin(psi);

            // set values on the message object
            PoseStamped.pose.position.x = posePositionX;
            PoseStamped.pose.position.y = posePositionY;
            PoseStamped.pose.position.z = posePositionZ;
            PoseStamped.pose.orientation.x = x;
            PoseStamped.pose.orientation.y = y;
            PoseStamped.pose.orientation.z = z;
            PoseStamped.pose.orientation.w = w;

            // Publish the message
            goal.publish(PoseStamped);
          };
    </script>
`
</details>
 
The input values appear to be published on the Message Object: 
[image]

minggangw, 2021-04-22:
Would you use [ros2cli](https://docs.ros.org/en/foxy/Concepts/About-Command-Line-Tools.html) to verify if the message has been sent practically?

newcanopies, 2021-04-22:
seems it is not sent practically
`ros2 topic list`
/clock
/parameter_events
/rosout
/robot/joint_states  _(urdf)_
/robot/set_joint_states _(urdf)_

@minggangw how to debug the above code and diagnose front vs backend? 
Where could transport be failing when` Rosbridge` logs "connected" and the `Message object` JSON logs correct input values?

newcanopies, 2021-04-22:
@minggangw 

above, the web-client HTML is served by ros2-web-bridge/bin/rosbridge.js

below, I tried serving the `web-client` via it's own websocket server (http/websocket8080)
and then connecting `ros2-web-bridge` in client mode `node bin/rosbridge.js --address ws://localhost:8080`

which logs the following:

`npm start web-clientserver.js`
```
Thu Apr 22 2021 17:48:44 GMT+0200 (Central   European Summer Time) New connection accepted from: http://localhost:8080 ::1 |   |   |   |   |   |   |   |  
-- | -- | -- | -- | -- | -- | -- | -- | --
Thu   Apr 22 2021 17:48:49 GMT+0200 (Central European Summer Time) Received   Message:   {"op":"advertise","id":"advertise:~/heartbeat:13","type":"heartbeat","topic":"~/heartbeat","latch":false,"queue_size":100}   from http://localhost:8080 ::1
Thu   Apr 22 2021 17:48:50 GMT+0200 (Central European Summer Time) Received   Message:   {"op":"publish","id":"publish:~/heartbeat:14","topic":"~/heartbeat","msg":{"alive":1},"latch":false}   from http://localhost:8080 ::1
```
`node bin/rosbridge.js --address ws://localhost:8080`
```
Thu   Apr 22 2021 17:49:29 GMT+0200 (Central European Summer Time) New connection   accepted from: undefined ::ffff:127.0.0.1 |   |   |   |   |   |   |   |  
Thu   Apr 22 2021 17:49:29 GMT+0200 (Central European Summer Time) Received   Message:   {"op":"status","level":"error","msg":"publish:   TypeError: Cannot read property 'get' of undefined"} from undefined   ::ffff:127.0.0.1
Thu   Apr 22 2021 17:49:29 GMT+0200 (Central European Summer Time) Received   Message:   {"op":"publish","id":"publish:~/heartbeat:22","topic":"~/heartbeat","msg":{"alive":1},"latch":false}   from http://localhost:8080 ::1
```
input values in web-client
```
Thu   Apr 22 2021 17:49:42 GMT+0200 (Central European Summer Time) Received   Message:   {"op":"publish","id":"publish:/move_base_simple/goal:3","topic":"/move_base_simple/goal","msg":{"header":{"seq":0,"stamp"
:{"secs":0,"nsecs":100},"frame_id":"map"},"pose":{"position":{"x":0,"y":0,"z":2.7},"orientation":{"x":0,"y":0,"z":0,"w":1}}},"latch":false}   from http://localhost:8080 ::1 |   |   |   |   |  
Thu   Apr 22 2021 17:49:42 GMT+0200 (Central European Summer Time) Received   Message:   {"op":"status","level":"error","msg":"publish:   TypeError: Cannot read property 'get' of undefined"} from undefined   ::ffff:127.0.0.1
Thu   Apr 22 2021 17:49:45 GMT+0200 (Central European Summer Time) Received   Message:   {"op":"publish","id":"publish:~/heartbeat:25","topic":"~/heartbeat","msg":{"alive":1},"latch":false}   from http://localhost:8080 ::1
Thu   Apr 22 2021 17:49:45 GMT+0200 (Central European Summer Time) Received   Message:   {"op":"status","level":"error","msg":"publish:   TypeError: Cannot read property 'get' of undefined"} from undefined   ::ffff:127.0.0.1
```

minggangw, 2021-04-23:
I noticed that when publishing a topic by the `ros2-web-bridge`, there is always an error

```
17:49:45 Received   Message:   {"op":"status","level":"error","msg":"publish:   TypeError: Cannot read property 'get' of undefined"} from undefined   ::ffff:127.0.0.1
```
I think that is why it failed. Please check if you have a field named `get` in the JSON object.

newcanopies, 2021-04-23:
> Please check if you have a field named `get` in the JSON object.

the only instance of "get" in the entire index.html is the camelcase getElementbyId 
posePositionX = Number(document.**get**ElementById('PositionX').value);

is this the correct command to run ros2-web-bridge as client?
`node bin/rosbridge.js --address ws://localhost:8080`

why does the rosbridge response logging:
17:49:29 New connection   accepted from: **undefined ::ffff:127.0.0.1**

to Publish a Topic/Message from browser-client to rosbridge, is the `subscribe` function necessary in the HTML?
```
goal.subscribe(function(message) {
          console.log(`Received message: ${message}`);
        });
```
would it preceede or follow `function pubMessage() {`

minggangw, 2021-04-23:
Do you get the these logs by runing `DEBUG=ros2-web-bridge* node bin/rosbridge.js`?

newcanopies, 2021-04-23:
@minggangw 

no, the terminal logs above are produced by [gazebo-bridge-server](https://github.com/osrf/gzweb/blob/master/gzbridge/server.js) of [gzweb](https://github.com/osrf/gzweb) (web-client)

when [gazebo-bridge-server](https://github.com/osrf/gzweb/blob/master/gzbridge/server.js) is running, 
trying to connect node bin/rosbridge.js --address ws://localhost:8080 in client mode
logs the unknown error:
17:48:44 New connection accepted from: http://localhost:8080 ::1  npm start [gazebo-bridge-server](https://github.com/osrf/gzweb/blob/master/gzbridge/server.js)
17:49:29 New connection accepted from: undefined ::ffff:127.0.0.1  node bin/rosbridge.js

minggangw, 2021-04-23:
If you are using `ros2-web-bridge` and want to check if the bridge works correctly, I suggest you run `DEBUG=ros2-web-bridge* node bin/rosbridge.js`

newcanopies, 2021-04-23:
`rosbridge_suite` "_default server uses autobahn, a python server implementation_"
what is the` ros2-web-bridge` default server?

newcanopies, 2021-04-23:
>  I suggest you run `DEBUG=ros2-web-bridge* node bin/rosbridge.js`

is this command correct to connect ros2-web-bridge to a running gazebo-bridge node?
`DEBUG=ros2-web-bridge* node bin/rosbridge.js --address ws://localhost:8080`

newcanopies, 2021-04-23:
hi @minggangw 

`DEBUG=ros2-web-bridge* node bin/rosbridge.js`

reveals what looks like a conflict connecting the two ros node-bridges on same localhost:8080

`node:   /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename   boost::detail::sp_member_access<T>::type   boost::shared_ptr<T>::operator->() const [with T =   gazebo::transport::Publication; typename   boost::detail::sp_member_access<T>::type =   gazebo::transport::Publication*]: Assertion `px != 0' failed.`

if `rosbridge_suite` "default server uses autobahn, a python server implementation"
what is the ros2-web-bridge default server?

minggangw, 2021-04-23:
Let me try to understand your problem, and I want to clarify the difference between rosbridge_suite and ros2-web-bridge first:

- [rosbridge_suite](https://github.com/RobotWebTools/rosbridge_suite): a general "bridge" designed for **ROS1** and works with [roslibjs](https://github.com/RobotWebTools/roslibjs) (front-end)
- [ros2-web-bridge](https://github.com/RobotWebTools/ros2-web-bridge): a general "bridge" designed for **ROS2** and also works with [roslibjs](https://github.com/RobotWebTools/roslibjs)

gzweb uses roslibjs to communicate with the bridge implemented by a nodejs [addon](https://github.com/osrf/gzweb/tree/master/gzbridge) if I understand correctly, the bridge is not rosbridge_suite or ros2-web-bridge. Does `gzweb` support ROS2 already?

newcanopies, 2021-04-23:
thank you for the definitions/clarifications @minggangw  (I am also reading all the available RobotWebTools documentation)

> gzweb uses roslibjs to communicate with the bridge implemented by a nodejs [addon](https://github.com/osrf/gzweb/tree/master/gzbridge) if I understand correctly, the bridge is not rosbridge_suite or ros2-web-bridge. Does `gzweb` support ROS2 already?

A) the native gzweb package is a Gazebo WebGL roslibjs client _only_. It has no UX command input-interface. 
the native gzweb package creates 
nodejs gazebo-bridge to Gazebo 
over a nodejs WebSocket server
as well as a static http server for the index.html

the result is Gazebo/robot loaded on localhost:8080

B) to validate controlling a robot in Gazebo using `rclnodejs`, 
I added to the gzweb index.html:
simple form <div> to input values to be mapped on the Message Object 
your script for creating a ROS Topic and Message, followed by Publish and Subscribe functions
(including required code to map values input in the browser html, to the created Message Object)
(the code maping browser input values to the Message Object, is already validated by correct JSON string logged in terminal)

C) as of a few weeks ago, gzweb works with ROS2 and can be compiled against node 12.20.1 which is also compatible with `rclnodejs` ^17

D) the boost error seen in `DEBUG=ros2-web-bridge* node bin/rosbridge.js` log 
is due to gzweb also creating a websocket server on localhost:8080

```
const WebSocketServer = require('./node_modules/websocket').server;
const http = require('http');
const fs = require('fs');
const path = require('path');
//const gzbridge = require('./build/Debug/gzbridge');

// HTTP server
let httpServer = http.createServer(staticServe);
httpServer.listen(port);

console.log(new Date() + " Static server listening on port: " + port);

// Websocket
let gzNode = new gzbridge.GZNode();
if (gzNode.getIsGzServerConnected())
{
  gzNode.loadMaterialScripts(staticBasePath + '/assets');
  gzNode.setPoseMsgFilterMinimumAge(0.02);
  gzNode.setPoseMsgFilterMinimumDistanceSquared(0.00001);
  gzNode.setPoseMsgFilterMinimumQuaternionSquared(0.00001);

  console.log('--------------------------------------------------------------');
  console.log('Gazebo transport node connected to gzserver.');
  console.log('Pose message filter parameters between successive messages: ');
  console.log('  minimum seconds: ' +
      gzNode.getPoseMsgFilterMinimumAge());
  console.log('  minimum XYZ distance squared: ' +
      gzNode.getPoseMsgFilterMinimumDistanceSquared());
  console.log('  minimum Quartenion distance squared:'
      + ' ' + gzNode.getPoseMsgFilterMinimumQuaternionSquared());
  console.log('--------------------------------------------------------------');
}
else
{
  materialScriptsMessage =
      gzNode.getMaterialScriptsMessage(staticBasePath + '/assets');
}

// Start websocket server
let wsServer = new WebSocketServer({
  httpServer: httpServer,
  // You should not use autoAcceptConnections for production
  // applications, as it defeats all standard cross-origin protection
  // facilities built into the protocol and the browser.  You should
  // *always* verify the connection's origin and decide whether or not
  // to accept it.
  autoAcceptConnections: false
});

wsServer.on('request', function(request) {

  // Accept request
  let connection = request.accept(null, request.origin);

  // If gzserver is not connected just send material scripts and status
  if (!gzNode.getIsGzServerConnected())
  {
    // create error status and send it
    let statusMessage =
        '{"op":"publish","topic":"~/status","msg":{"status":"error"}}';
    connection.sendUTF(statusMessage);
    // send material scripts message
    connection.sendUTF(materialScriptsMessage);
    return;
  }

  connections.push(connection);

```

To integrate gzweb into ros2-web-bridge, I added the above into ros2-web-bridge Index.js along with installing all node_modules.

I am not sure if either of the WebSocket servers can support multiple clients, 
or if gzweb and ros2-web-bridge can not simultaneously share a websocket.

This is the reason the Message is not being practically published.

newcanopies, 2021-04-23:
That's right. Thank you - the boost:: websocket conflict was resolved by:

running web-client http+websocket on 8080
scripting `bridgePort` inline-html @ 9090
running ros2-web-bridge on 9090

the published Topic is live in ROS, but now ros2-web-bridge throws a _TypeError: Invalid argument:_

newcanopies, 2021-04-24:
All correct and Yes it does! @minggangw 

I was able to **integrate** the two nodejs based packages, and resolve some boost::transport errors by:
running gzweb-cli to Gazebo websocket server @8080
running `ros2-web-bridge` @9090
publishing to bridgePort @9090 
 
ROS2 introspection echoes the Topic correctly, and the below debug log shows no errors -- however Gazebo control is not established -- what else could be failing in transport???

`DEBUG=ros2-web-bridge* npm start`

> web-cli@0.2.0 start /home/user/ros2-web-bridge-hybrid
> if [ $npm_config_port ]; then port=$npm_config_port; fi; cd gzbridge && ./server.js $port

Sat Apr 24 2021 14:32:40 GMT+0200 (Central European Summer Time) Static server listening on port: 8080
--------------------------------------------------------------
Gazebo transport node connected to gzserver.
Pose message filter parameters between successive messages: 
  minimum seconds: 0.02
  minimum XYZ distance squared: 0.00001
  minimum Quartenion distance squared: 0.00001
--------------------------------------------------------------
**Websocket started on ws://localhost:9090**                   <-- index.js:87  options.port = options.port || 9090;

  ros2-web-bridge:Bridge Status level set to error (0) +0ms
  ros2-web-bridge:**Bridge Web bridge** d8516481-f33c-4694-9646-cb8c3c1ee695 is created +1ms
Sat Apr 24 2021 14:32:54 GMT+0200 (Central European Summer Time) New connection accepted from: http://localhost:8080 ::1

**//Gazebo-bridge**
14:32:55  Received Message: {"op":"subscribe","id":"subscribe:~/status:1","type":"status","topic":"~/status","compression":"none","throttle_rate":0,"queue_length":0} from http://localhost:8080 ::1
14:32:55  Received Message: {"op":"subscribe","id":"subscribe:~/material:2","type":"material","topic":"~/material","compression":"none","throttle_rate":0,"queue_length":0} from http://localhost:8080 ::1
14:32:55  Received Message: {"op":"subscribe","id":"subscribe:~/scene:3","type":"scene","topic":"~/scene","compression":"none","throttle_rate":0,"queue_length":0} from http://localhost:8080 ::1
 14:32:55  Received Message: {"op":"subscribe","id":"subscribe:~/physics:4","type":"physics","topic":"~/physics","compression":"none","throttle_rate":0,"queue_length":0} from http://localhost:8080 ::1
14:32:55 Received Message: {"op":"subscribe","id":"subscribe:~/pose/info:5","type":"pose","topic":"~/pose/info","compression":"none","throttle_rate":0,"queue_length":0} from http://localhost:8080 ::1
14:32:55  Received Message: {"op":"subscribe","id":"subscribe:~/request:6","type":"request","topic":"~/request","compression":"none","throttle_rate":0,"queue_length":0} from http://localhost:8080 ::1
14:32:55 Received Message: {"op":"subscribe","id":"subscribe:~/model/info:7","type":"model","topic":"~/model/info","compression":"none","throttle_rate":0,"queue_length":0} from http://localhost:8080 ::1
14:32:55 Received Message: {"op":"subscribe","id":"subscribe:~/visual:8","type":"visual","topic":"~/visual","compression":"none","throttle_rate":0,"queue_length":0} from http://localhost:8080 ::1
14:32:55 Received Message: {"op":"subscribe","id":"subscribe:~/world_stats:9","type":"world_stats","topic":"~/world_stats","compression":"none","throttle_rate":0,"queue_length":0} from http://localhost:8080 ::1
14:32:55 Received Message: {"op":"subscribe","id":"subscribe:~/factory/light:10","type":"light","topic":"~/factory/light","compression":"none","throttle_rate":0,"queue_length":0} from http://localhost:8080 ::1
14:32:55 Received Message: {"op":"subscribe","id":"subscribe:~/light/modify:11","type":"light","topic":"~/light/modify","compression":"none","throttle_rate":0,"queue_length":0} from http://localhost:8080 ::1
14:32:55 Received Message: {"op":"call_service","id":"call_service:~/roads:12","service":"~/roads","type":"roads","args":{"name":"roads"}} from http://localhost:8080 ::1
14:32:55 Received Message: {"op":"unsubscribe","id":"subscribe:~/scene:3","topic":"~/scene"} from http://localhost:8080 ::1
14:33:00 Received Message: {"op":"advertise","id":"advertise:~/heartbeat:13","type":"heartbeat","topic":"~/heartbeat","latch":false,"queue_size":100} from http://localhost:8080 ::1
14:33:00 Received Message: {"op":"publish","id":"publish:~/heartbeat:14","topic":"~/heartbeat","msg":{"alive":1},"latch":false} from http://localhost:8080 ::1
14:33:05b Received Message: {"op":"publish","id":"publish:~/heartbeat:15","topic":"~/heartbeat","msg":{"alive":1},"latch":false} from http://localhost:8080 ::1

**//rosBridge**
  ros2-web-bridge:**Bridge JSON command received**: {"op":"**advertise**","id":"advertise:/move_base_simple/goal:2","type":"geometry_msgs/PoseStamped","topic":"/move_base_simple/goal","latch":false,"queue_size":100} +12s
  ros2-web-bridge:**Bridge advertise a topi**c: /move_base_simple/goal +1ms
  ros2-web-bridge:**ResourceProvider Publisher has been created**, and the topic name is /move_base_simple/goal. +0ms
  ros2-web-bridge:Bridge Suppressed: {"op":"status","level":"none","msg":"OK","id":"advertise:/move_base_simple/goal:2"} +2ms
  ros2-web-bridge:**Bridge JSON command received**: {"op":"publish","id":"publish:/move_base_simple/goal:3","topic":"/move_base_simple/goal","msg":{"header":{"seq":0,"stamp":{"sec":0,"nanosec":100},"frame_id":"map"},"pose":{"position":{"x":0,"y":0,"z":0.6},"orientation":{"x":0,"y":0,"z":0,"w":1}}},"latch":false} +1ms
[... cut ...]
  ros2-web-bridge:Bridge Suppressed: {"op":"status","level":"none","msg":"OK","id":"publish:/move_base_simple/goal:3"} +4ms
  ros2-web-bridge:**Bridge Send message to subscription**. +7ms

**//Gazebo-bridge**
Sat Apr 24 2021 14:33:10 GMT+0200 (Central European Summer Time) Received Message: {"op":"publish","id":"publish:~/heartbeat:16","topic":"~/heartbeat","msg":{"alive":1},"latch":false} from http://localhost:8080 ::1
^C
Gracefully shutting down from SIGINT (Ctrl-C)
^CTerminated
```

![image](https://user-images.githubusercontent.com/65022111/115959554-0e6b4100-a50d-11eb-96f5-ec13e8421ad4.png)

newcanopies, 2021-04-24:
As a simpler question, how to Publish from `ros2-web-bridge` to `Gazebo11` running as headless `gzserver`?

newcanopies, 2021-04-24:
Publishing JointStates from terminal moves the Gazebo robot:
ros2 topic pub /robot/set_joint_states trajectory_msgs/msg/JointTrajectory '{header: {frame_id: world}, joint_names: {joint_name}, points: [{positions: {-2}}]}'

and rosbridge server logs it too: 
```
  ros2-web-bridge:Bridge Send message to subscription. +1s
  ros2-web-bridge:Bridge Send message to subscription. +999ms 
```

Publishing PoseStamped from terminal **does not move** Gazebo robot 
ros2 topic pub /move_base_simple/goal geometry_msgs/PoseStamped '{ header: { frame_id:  "map"}, pose: { position: { x: 0.2, y: 0 }, orientation: { x: 0, y: 0, z: 0, w: 1 } } }'

however rosbridge server logs it: 
```
  ros2-web-bridge:Bridge Send message to subscription. +1s
  ros2-web-bridge:Bridge Send message to subscription. +999ms 
```

What should be the PoseStamped `frame_id` in ROS2 ?
"world"  
"map"
"base_link"
"end_effector_link"  

Is a KDL URDF parser (ROS1 only) required to control a Gazebo robot with PoseStamped message?

and RosBridge logs 
```
  ros2-web-bridge:Bridge Send message to subscription. +1s
  ros2-web-bridge:Bridge Send message to subscription. +998ms
  ros2-web-bridge:Bridge Send message to subscription. +1s
```

minggangw, 2021-04-25:
I'm not clear about your entire scenario, but it seems that you use some ROS1 component?

newcanopies, 2021-04-25:
> 
> 
> I'm not clear about your entire scenario, but it seems that you use some ROS1 component?

`roslibjs` is the only ROS1 component. To your knowledge, should PoseStamped move a robot without any additional interfaces or packages like moveit?

minggangw, 2021-04-26:
**ros2-web-bridge** is compatible with roslibjs by complying with [rosbridge v2 protocol](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md)

I'm not familiar with Gazebo simulator, but I deem publishing a `PoseStamped` topic is enough to control your virtual robot. Would you have a brief graph for your entire framework? It seems that you use the gzweb bridge and ros2-web-bridge simultaneously?

newcanopies, 2021-04-26:
thanks for your insight @minggangw 
it would be great if just adding an equivalent of the  libgazebo_ros_joint_pose_trajectory.so plugin to the robot.sdf would interface it with PoseStamped, but it is starting to look like moveit2 package may be required?

since I integrated gazebo-client-bridge into the `ros2-web-bridge` package, it is implicit under the ros2-web-bridge in the graph:
[image]
[image]

newcanopies, 2021-04-26:
here is the graph with gazebo and ros2-web-bridge running and publishing from terminal:
`ros2 topic pub /move_base_simple/goal geometry_msgs/PoseStamped '{ header: { frame_id:  "world"}, pose: { position: { x: 0.4, y: 0.7 }, orientation: { x: 0, y: 0, z: 0, w: 1 } } }'`
[image]

minggangw, 2021-04-26:
I'm not clear why do you need a `ros2-web-bridge` here. For my understanding, GzWeb's NodeJS [server](https://github.com/osrf/gzweb/blob/master/gzbridge/server.js) will work as a bridge between the Javascript and gazebo C++ side. So, what is the role of `ros2-web-bridge`, I think the whole system is running on ROS2.

newcanopies, 2021-04-26:
> why do you need a `ros2-web-bridge` here

Not sure, good question, isn't `ros2-web-bridge`'s node_module `rclnodejs` required to publish PoseStamped?

minggangw, 2021-04-27:
> isn't ros2-web-bridge's node_module rclnodejs required to publish PoseStamped?

Yes, `rclnodejs` acts as back-end of `ros2-web-bridge` to access the ROS2, e.g. pub/sub a topic, but it cannot connect to the gzserver. A general usage is below:

[image]

newcanopies, 2021-04-27:
> but it cannot connect to the gzserver.

OK, that is exactly why I try to integrate GzWeb to additionally "bridge" the Publishing to gzserver. 

Does that mean that `ros2-web-bridge` is for controlling hardware robots via web, 
as not for controlling simulated robots in Gazebo?

When GzWeb and ros2-web-bridge share the same PORT a boost Gazebo::transport error is thrown:
_node: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type = gazebo::transport::Publication*]: Assertionpx != 0' failed_

newcanopies, 2021-05-01:
hi @minggangw 
still racking my brains on this. 
is this the missing interface for Gazebo to receive the web-cli published message? https://github.com/RobotWebTools/tf2_web_republisher
what would it take to adapt it to ROS2?

thank you!

minggangw, 2021-05-03:
To adapt the `tf2_web_republisher` to ROS2, I think you have to re-implement it, which needs a lot of work. To identify if the `ros2-web-bridge` works correctly, I suggest you could change the topic type of [pub](https://github.com/RobotWebTools/ros2-web-bridge/blob/develop/examples/html/publisher.html) and [sub](https://github.com/RobotWebTools/ros2-web-bridge/blob/develop/examples/html/subscription.html) to `PoseStamped `, then you can check if the topic can be received.

newcanopies, 2021-05-27:
> change the topic type of [pub](https://github.com/RobotWebTools/ros2-web-bridge/blob/develop/examples/html/publisher.html) and [sub](https://github.com/RobotWebTools/ros2-web-bridge/blob/develop/examples/html/subscription.html) to `PoseStamped `, then you can check if the topic can be received.

The topic is published in the bash terminal, and can be echoed in the bash terminal, but the message is not displayed in HTML
Is` ros2-web-bridge` able to receive this topic?
Publisher
[image]
https://gist.github.com/newcanopies/85ccc01a27a60a82fccd0fd7a1d7398e
Subscriber
[image]

https://gist.github.com/newcanopies/f679c3ad8464534d29ed5690d56a0532

newcanopies, 2021-05-27:
Also, the following error is printed after `Ctrl-C`

```
ros2-web-bridge/node_modules/rclnodejs/lib/publisher.js:59
      rclnodejs.publish(this._handle, rawMessage);
                ^

Error: publisher pointer is invalid, at /tmp/binarydeb/ros-foxy-rcl-1.1.11/src/rcl/publisher.c:434
    at Publisher.publish (/home/user/ros2-web-bridge/node_modules/rclnodejs/lib/publisher.js:59:17)
    at Timeout._onTimeout (/home/user/ros2-web-bridge/examples/index_posestamped.js:162:17)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
```

newcanopies, 2021-05-27:
Ending index_rclnode.js 

gives this error:

```
ros2-web-bridge/node_modules/rclnodejs/lib/publisher.js:59
      rclnodejs.publish(this._handle, rawMessage);
                ^

Error: publisher pointer is invalid, at /tmp/binarydeb/ros-foxy-rcl-1.1.11/src/rcl/publisher.c:434
    at Publisher.publish (/home/user/ros2-web-bridge/node_modules/rclnodejs/lib/publisher.js:59:17)
    at Timeout._onTimeout (/home/user/ros2-web-bridge/examples/index_rclnode.js:24:17)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
```

ros2-web-bridge/examples/index_rclnode.js:24:17)

    setInterval(function () {
      
      
              publisher.publish({
      
 
                <span class="pl-c1">header</span>: <span class="pl-kos">{</span>

newcanopies, 2021-06-07:
hi @minggangw 
do you think there is an error in below scripts or bridge?

$ node examples/[index_rclnode.js
](https://gist.github.com/newcanopies/976387554fe2504e232b8e8efd18a42a)

[Publisher](https://gist.github.com/newcanopies/85ccc01a27a60a82fccd0fd7a1d7398e)

[Subscriber](https://gist.github.com/newcanopies/f679c3ad8464534d29ed5690d56a0532
image)