RobotWebTools/ros2-web-bridge #181: rosbridge can not be terminated from bash

newcanopies, 2021-04-12:
Hello, 

thank you for this great package. 

is there a reason rosbridge node can not be easily terminated in bash terminal?
even after closing the browser window, terminal remains hung-up 
[image]

christophebedard, 2021-04-12:
I also have this issue. I have to manually kill it (e.g. using htop).

Maybe something like this could be an easy fix: https://stackoverflow.com/a/53535316

newcanopies, 2021-04-21:
> Maybe something like this could be an easy fix: https://stackoverflow.com/a/53535316

thanks @christophebedard 
it does free-up that terminal, but the rosbridge node remains running in the background, listening on the PORT EADDRINUSE

christophebedard, 2021-04-21:
I assume one of these has to be called inside the signal handling function: `shutDown()` or `closeAllBridges()` (in `index.js`).

newcanopies, 2021-04-21:
thank you 

tried all the variations, so far no luck, 
SIGINIT logs the graceful shutdown message 
but rosbridge node remains running

newcanopies, 2021-04-21:
@christophebedard is it even possible to call inside the functions?
[image]

newcanopies, 2021-04-21:
@christophebedard 

none seem to work:
<details>
`
// Copyright (c) 2017 Intel Corporation. All rights reserved.
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at
//
//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

'use strict';

const rclnodejs = require('rclnodejs');
const WebSocket = require('ws');
const Bridge = require('./lib/bridge.js');
const debug = require('debug')('ros2-web-bridge:index');

// rclnodejs node
let node;

// Websocket server (or client if client mode set via --address)
let server;
let connectionAttempts = 0;

// Map of bridge IDs to Bridge objects
let bridgeMap = new Map();

function closeAllBridges() {
  process.on('SIGINT', () => process.exit(1));
  bridgeMap.forEach((bridge, bridgeId) => {
    bridge.close();
  });
}

function shutDown(error) {
  process.on('SIGINT', () => process.exit(1));
  // Closing the server triggers the individual connections to be closed.
  if (server) {
    server.close();
  }
  if (!rclnodejs.isShutdown()) {
    rclnodejs.shutdown();
    process.on('SIGINT', () => process.exit(1));
  }
  if (error) {
    throw error;
  }
}

function createServer(options) {
  options = options || {};
  options.address = options.address || null;
  process.on('exit', () => {
    debug('Application will exit.');
    shutDown(
      process.on('SIGINT', () => process.exit(1))
    );
    process.on('SIGINT', () => process.exit(1));
  });
  return rclnodejs.init()
    .then(() => {
      node = rclnodejs.createNode('ros2_web_bridge');
      rclnodejs.spin(node);
      debug('ROS2 node started');
      createConnection(options);
    })
    .catch(error => shutDown(error));
}

function createConnection(options) {
  if (options.address != null) {
    debug('Starting in client mode; connecting to ' + options.address);
    server = new WebSocket(options.address);
  } else {
    options.port = options.port || 8080;
    //options.port = options.port || 9090;
    debug('Starting server on port ' + options.port);
    server = new WebSocket.Server({port: options.port});
  }

  const makeBridge = (ws) => {
    let bridge = new Bridge(node, ws, options.status_level);
    bridgeMap.set(bridge.bridgeId, bridge);

    bridge.on('error', (error) => {
      debug(`Bridge ${bridge.bridgeId} closing with error: ${error}`);
      bridge.close();
      bridgeMap.delete(bridge.bridgeId);
    });

    bridge.on('close', (bridgeId) => {
      bridgeMap.delete(bridgeId);
    });
  };

  server.on('open', () => {
    debug('Connected as client');
    connectionAttempts = 0;
  });
  
  if (options.address) {
    makeBridge(server);
  } else {
    server.on('connection', makeBridge);
  }

  server.on('error', (error) => {
    closeAllBridges(
      process.on('SIGINT', () => process.exit(1))
      );
    debug(`WebSocket error: ${error}`);
  });

  server.on('close', (event) => {
    debug(`Websocket closed: ${event}`);
    if (options.address) {
      closeAllBridges(
        process.on('SIGINT', () => process.exit(1))
      );
      connectionAttempts++;
      // Gradually increase reconnection interval to prevent
      // overwhelming the server, up to a maximum delay of ~1 minute
      // https://en.wikipedia.org/wiki/Exponential_backoff
      const delay = Math.pow(1.5, Math.min(10, Math.floor(Math.random() * connectionAttempts)));
      debug(`Reconnecting to ${options.address} in ${delay.toFixed(2)} seconds`);
      setTimeout(() => createConnection(options), delay*1000);
    }
  });

  let wsAddr = options.address || `ws://localhost:${options.port}`;
  console.log(`Websocket started on ${wsAddr}`);
}

module.exports = {
  createServer: createServer,
};

`
</details>

newcanopies, 2021-04-21:
this works for me
ros2-web-bridge/index.js:**136**
[image]

newcanopies, 2021-04-21:
thank you, this works
### ros2-web-bridge/index.js:136
[image]

closed 2021-04-21 by newcanopies

minggangw, 2021-04-25:
Please checkout the latest [v0.3.1](https://www.npmjs.com/package/ros2-web-bridge/v/0.3.1) which has this fix, thanks!