RobotWebTools/ros2-web-bridge #187: Latest rclnodejs version 0.19.0 doesn't work on Foxy

ManuelZ, 2021-05-26:
I'm using Foxy and installing ros2-web-bridge, which depends on rclnodejs version 0.x.1, which lately means installing 0.19.0, which doesn't work on Foxy. It brings up an error as stated here:
https://github.com/RobotWebTools/rclnodejs/issues/792

What could be done to allow a Foxy user to install ros2-web-bridge without trouble?

Aposhian, 2021-05-26:
I think that ros2-web-bridge 0.3 should lock down the rclnodejs dependency to ^0.18.x and bump the version for galactic to 0.4. How is this project going to manage versions for different ROS2 releases?

emersonknapp, 2021-05-26:
That seems like a good strategy to me - perhaps we should note the version-mappingin the README as well - a simple table like:

* Foxy: use rclnodejs ^0.18.x
* Galactic: use rclnodejs ^0.19.x
* Rolling: always use latest

minggangw, 2021-05-27:
This issue has been fixed by https://github.com/RobotWebTools/rclnodejs/commit/ab69738d8978f37549bec36ba5c6fd8838b5b243, and you can find which version matches a specific ROS release at https://github.com/RobotWebTools/rclnodejs#installing-rclnodejs

minggangw, 2021-05-31:
Please upgrade the rclnodejs to [0.19.1](https://www.npmjs.com/package/rclnodejs/v/0.19.1) that has resolved the incompatibility problem on Galactic.