RobotWebTools/ros2-web-bridge #195: Using Web-Bridge as part of a React application

eden-desta, 2021-09-07:
Hi! 
I have used this before as a standalone, while running the steps provided in the readME. I am interested in understanding if it can be packed as part of a React application to start up when we call `npm start`. I am a newbie with React so this might not be the right terminology so to speak but any input would be great!

Thank you! 
And thanks for providing this brilliant lib!

raryant, 2021-09-16:
yes, React is frontend framework, if you using npm start it will start in development mode, you should start the web bridge using service. CMIIW

flynneva, 2021-09-16:
@eden-desta if you're looking for inspiration feel free to build a ROS2 version of my ROS1 react-ros package: https://github.com/flynneva/react-ros

I've also tried a go at integrating react-three-fiber (aka three.js) as a separate package (react-ros-three). Feel free to look at that as well and use it as a template if you want to make your own!

sandeshworld, 2021-10-01:
These two things might help you: 
https://github.com/RobotWebTools/roslibjs/issues/252#issuecomment-306202003
https://github.com/RobotWebTools/roslibjs/issues/252#issuecomment-883302274

It helped me use it on for my React Application.

closed 2021-12-05 by eden-desta